casper-astro/mlib_devel #86: Version Info on the front page of readthedocs

jack-h, 2019-06-27:
As far as I can tell there is _no_ information in this repository explaining which versions of software are needed for which platform.

AdamI75, 2019-06-27:
Yes, that is true. It is on the tutorials_devel readthedocs and on the Berkeley "getting_started" link though.

jack-h, 2019-06-27:
Yeah, I think if the mlib_devel instructions are going to include an
install guide, they should probably directly contain instructions on which
versions to use :)

bjbford, 2019-06-27:
@jack-h agreed, I totally thought our version table was there already... We should just duplicate this info: https://casper-tutorials.readthedocs.io/en/latest/#environment-setup

jack-h, 2019-06-28:
Yes. :)

jack-h, 2019-07-18:
PR #99 puts this at the front of the installation instructions

closed 2020-10-16 by jkocz