casper-astro/mlib_devel #126: AXI write interface liable to fail if user_clk < axi_clk

jack-h, 2020-05-28:
The clock-domain crossing double-registers used to connect AXI registers to the user's simulink design are liable to miss write events if the axi clock is faster than the simulink clock (user_clk) since writable registers from the AXI interconnect subsystem only hold the IP_BUS_VALID signal high for one cycle.

Should there be some more sophisticated handshaking (as in the WB software registers)?

jack-h, 2020-05-28:
@AdamI75 -- FYI^. I discovered this when using the ADM-PCIE-9H7 with a 125 MHz AXI clock generated by the PCI core, along with a user_clk=sys_clk=100MHz. For now I'm avoiding the issue by turning the AXI clock down to 62.5 MHz. I'm not sure what the standard configuration is with the other AXIfied platforms, and whether this is going to be an issue for them too.

wnew, 2020-05-29:
@AdamI75 FTI. Ill take a look at this next week.

AdamI75, 2020-06-01:
@jack-h and @wnew. Ahhh, yes. We typically use a much lower clock frequency (user_clk > axi_clk) for AXI comms. It also assists with timing closure. At the moment I am just suing AXI for comms, but at some point when we stream then we may need to up the clock frequency for streaming interfaces. Although streaming interfaces will not use this IP_BUS_VALID signal in my mind. I am not sure if it is worth changing it now since a lower clock frequency is desirable, in my opinion. We can add this as a requirement to the future toolflow optimisation in the mean time?