casper-astro/mlib_devel #139: Default yaml loader issue

cain986, 2021-02-08:
I ran into an issue when building a Casper project using the `exec_flow.py` script. I was able to resolve it, but I thought I'd bring this to your attention. Here is the issue:
```
/data2/development/mlib_devel/jasper_library/castro.py:32: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  c = yaml.load(fh)
Traceback (most recent call last):
  File "/data2/development/mlib_devel/jasper_library/exec_flow.py", line 222, in <module>
    backend.import_from_castro(backend.compile_dir + '/castro.yml')
  File "/data2/development/mlib_devel/jasper_library/toolflow.py", line 1089, in import_from_castro
    self.castro = castro.Castro.load(filename)
  File "/data2/development/mlib_devel/jasper_library/castro.py", line 32, in load
    c = yaml.load(fh)
  File "/data2/development/.virtualenv/casper_venv_dcain/lib/python3.6/site-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/data2/development/.virtualenv/casper_venv_dcain/lib/python3.6/site-packages/yaml/constructor.py", line 51, in get_single_data
    return self.construct_document(node)
  File "/data2/development/.virtualenv/casper_venv_dcain/lib/python3.6/site-packages/yaml/constructor.py", line 55, in construct_document
    data = self.construct_object(node)
  File "/data2/development/.virtualenv/casper_venv_dcain/lib/python3.6/site-packages/yaml/constructor.py", line 100, in construct_object
    data = constructor(self, node)
  File "/data2/development/.virtualenv/casper_venv_dcain/lib/python3.6/site-packages/yaml/constructor.py", line 429, in construct_undefined
    node.start_mark)
yaml.constructor.ConstructorError: could not determine a constructor for the tag 'tag:yaml.org,2002:python/object:castro.Castro'
```

This resolved it for me:
```diff
--- a/mlib_devel/jasper_library/castro.py-BAK
+++ b/mlib_devel/jasper_library/castro.py
@@ -29,7 +29,7 @@ class Castro(object):
         loads this class object from a yaml file and assert that it is of type Castro
         '''
         with open(filename, 'r') as fh:
-            c = yaml.load(fh)
+            c = yaml.load(fh, Loader=yaml.Loader)
             assert isinstance(c, Castro)
             return c
```

AdamI75, 2021-02-09:
Hi @cain986, we are aware of this issue and have submitted a fix on the casper-astro/mlib_devel/ (casper-astro-soak-test branch). This will eventually get merged to the master branch. Try work off the casper-astro-soak-test branch for now or go ahead with your modification.

Thanks for brining this to my attention.

Kind regards, 

Adam Isaacson

jack-h, 2021-02-09:
I've just cherry-picked this fix and raised a PR against the master branch
of casper-astro, if someone with push access feels the urge to approve and
merge.

Thanks for tracing this down @cain986, and thanks for fixing in the soak
branch, Adam.

AdamI75, 2021-02-09:
Okay, thanks @jack-h. I will approve and merge now.

AdamI75, 2021-02-09:
@jack-h and @cain986: I have merged into the master branch, so this issue can be closed. @jack-h: thanks for cherry picking my previous commit :).

AdamI75, 2021-02-09:
This issue can be closed now - it has been merged into casper-astro/mlib_devel (master and casper-astro-soak-test-branch).

closed 2021-02-09 by AdamI75