casper-astro/mlib_devel #141: exec_flow partial compiles

realtimeradio, 2021-03-18:
`exec_flow.py` has flags which enable you to turn off bits of the compile. They [probably] work if you want to stop later stages of the compile from running, but don't work at all if you want to (eg) regenerate an FPG file, but not rerun `--backend`.
It would be nice if this weren't the case. One use case, which I just encountered, is if a user wants to intervene in a compile to tweak timing issues in vivado and then merge the new bitstream into the fpg.

AdamI75, 2021-03-19:
@realtimeradio have you checked out mlib_devel/jasper_library/utils/make_fpg.py. This allows you to create an fpg at anytime provided you have already done one build. You use it like this "python make_fpg -d <path to test_models build folder> -i 1." I have tested it for SKARAB and it works well e.g. testing a new elf without compiling again. The "-i 1" indicates that it will generate the fpg and use the bit file from "impl1". This may need to be modified for other boards, but is a very useful function. I suppose we should document this.

realtimeradio, 2021-03-19:
Interesting -- I didn't know about this. Thanks for the pointer!

closed 2021-03-19 by realtimeradio