casper-astro/mlib_devel #176: pfb_fir_real no longer accepts parameters

kiranshila, 2022-09-14:
Since updating to the m2021a, setting any of the parameters in the `pfb_fir_real` block doesn't seem to actually do anything. I noticed this after a call to `update_casper_blocks` and it was no longer the proper size in my model.

kiranshila, 2022-09-15:
It seems the library object for this block is missing its initialization script
[image]

mitchburnett, 2022-09-15:
#177 addresses this with some other fixes that were found to be wrong with the complex `fft` and general `pfb_fir`

jkocz, 2022-09-16:
Closing this as #177 has been merged. Thanks @kiranshila and @mitchburnett!

closed 2022-09-16 by jkocz