casper-astro/mlib_devel #178: Compilation of designs with no AXI bram / software registers fail

SergeyMushinsky, 2022-09-16:


AdamI75, 2022-09-16:
@jkocz I had forgotten about this. Is this something we would like resolved, in your opinion? I am sure a plan could be made. Let me know.

jack-h, 2022-09-16:
I _think_ one relatively easy improvement would be to add a `drc` method (
https://github.com/casper-astro/mlib_devel/blob/bc2bc99c27793ead0cfddf4c9c399624dfde0f98/jasper_library/yellow_blocks/yellow_block.py#L202)
to
https://github.com/casper-astro/mlib_devel/blob/m2021a/jasper_library/yellow_blocks/axi4lite_interconnect.py

It looks like the axi interconnect yellow block should know enough via it's
`memory_map` attribute to flag if these aren't enough BRAMs / regs.
Obviously this wouldn't fix the underlying issue, but at least the user
could be told something useful about what is wrong rather than the current
cryptic messages