casper-astro/mlib_devel #185: JAM Indexing Errors Prevents Compilation

kiranshila, 2022-10-21:
As far as I (and the C compiler) can tell, the constant `CORE_INFO` is a single byte in some but not all cases:

https://github.com/casper-astro/mlib_devel/blob/1109264276b6ff0209e03c1bc69c0a64577772dc/jasper_library/sw/jam/casper_tapcp.c#L244-L246

So in lines such as

https://github.com/casper-astro/mlib_devel/blob/1109264276b6ff0209e03c1bc69c0a64577772dc/jasper_library/sw/jam/casper_tapcp.c#L1185

and

jack-h, 2023-05-14:
what Xilinx tool version are you compiling with?

david-macmahon, 2023-05-14:
Maybe something has changed in the toolchain related to the linker?  More details on how `_core_info` are intended to (and used to) work can be found in #9f4dd46.

david-macmahon, 2023-05-14:
Not sure why GitHub didn't auto-link that commit hash to this:
https://github.com/casper-astro/mlib_devel/commit/9f4dd462c37580d7f4fd267c36a667b69ad285be

kiranshila, 2023-05-14:
> what Xilinx tool version are you compiling with?

Not sure - I think this was 2021? It was a while ago when I tried :(