casper-astro/mlib_devel #193: [Easy] No recursive cloning in tutorial instructions

jwkunz, 2023-01-10:
I was getting a failed jasper synthesis on the vanilla zcu111_tut_onehundred_gbe file.  I went digging through the generate source code and after some splunking in my file system I realized this issue was a missing  kutleng_skarab2_bsp_firmware directory.  Looking in the github repo, I realized this was a submodule that was not in my local filesystem.

The current tutorial instructions do not mention the recursive cloning of submodules.  I recommend adding this to the instructions for new users.

In summary,

Change:
"git clone https://github.com/casper-astro/mlib_devel.git"
To:
"git clone https://github.com/casper-astro/mlib_devel.git --recursive"

in the "Getting Started With RFSoC" page.  May also add a note about this in the text.

mitchburnett, 2023-01-10:
This is in the `Getting Started With RFSoC` page under `Toolflow Setup` for initialization of submodules

https://casper-toolflow.readthedocs.io/projects/tutorials/en/latest/tutorials/rfsoc/tut_getting_started.html#toolflow-setup

closed 2023-01-10 by mitchburnett

jwkunz, 2023-01-10:
Silly me.  I scanned over that.