casper-astro/mlib_devel #199: RFDC block stops working if .slx file is renamed

gallicchio, 2023-06-02:
If I rename the .slx file or copy it to a new file to make changes, the tutorial [rfsoc4x2_tut_rfdc_real.slx](https://github.com/casper-astro/tutorials_devel/blob/main/rfsoc/tut_rfdc/rfsoc4x2_tut_rfdc_real.slx) still builds without errors, but the ADCs only return -1.

I went into the BLOCK toolbar and click Expand on the rfdc block. The gateways that are revealed do include the name of the original file, not the renamed file.

mitchburnett, 2023-10-06:
Hopefully this will be resolved in #205 that may be merging shortly. Once that is, see this this issue goes a way for you. Copying an already configured RFDC Simulink yellow block into a new .slx did have issues however the toolflow build would not complete correctly. A work around was to just open the mask, change a value, and then change it back. This caused the mask to redraw itself, including redrawing the underlying ports. This logic to redraw when the state of the block has changed, including copying to a new slx, is improved #205. The "ADCs only return -1" is a subtle nuance that may need to be investigated further providing more information.

mitchburnett, 2024-05-15:
If the "ADCs only return -1" please open a new issue as this may be a slightly different issue.

closed 2024-05-15 by mitchburnett