casper-astro/mlib_devel #200: Test models out of date

jkocz, 2023-06-10:
It looks like may of the test models in the "jasper_library/test_models" directory have not been updated in 5 years. 

In particular, some of them refer to scripts / black boxes that I am not able to find elsewhere in the code.  We either need to add this code (tagging @AdamI75 as the test models seemed to mostly come from SARAO) or remove those that no longer work.

AdamI75, 2023-06-11:
@jkocz Thanks for the heads up - 5 years seems a bit long not to have updated the "jasper_library/test_models".  That would have been 2018 and that would have been when we were in China busy with SKARAB. It could be that casper-astro is missing some pull requests from the ska-sa repo? I will check it out next week and let you know, thanks - busy debugging hardware infrastructure at the moment. I can imagine that it would either be SKARAB, Red Pitaya and/or Alveo test models that have not been updated. My understanding is that SKARAB should definitely be up to date when it comes to R2018a and Vivad 2019.1.1. Red Pitaya should "mostly" be up to date. There may be some Alveo test models that need to be up to date, but I was also under the impression this was done. Anyway, I will let you know.

AdamI75, 2023-06-14:
I have spoken with @wnew and he says there may be some slx files that are repeats of tutorials and some slx files that can be removed. He will check and then I will check when he is done. We will issue a pull request to casper-astro when done, thanks. @jkocz please can you give me a time frame of when this needs to be complete by as @wnew and I have other research deadlines to consider? Thanks!

jkocz, 2023-06-14:
@AdamI75 Thank you! I don't think there is a huge rush. My current plan is to try and get this fixed to the m2021a-dev branch as we can over the next few weeks, and then merged into the main branch prior to the workshop this year.

AdamI75, 2023-06-15:
@jkocz Thanks for letting me know. Wesley and I will get on this as soon as we are able to. We will do a pull request to m2021a-dev when ready. It shouldn't be too bad, I hope!

AdamI75, 2023-06-19:
@jkocz I will start looking at this from next week Monday after I have completed my deadline on Friday. It is just a question of allocating time.

AdamI75, 2023-07-05:
@jkocz Wesley and I have cleaned this up - please see pull request https://github.com/casper-astro/mlib_devel/pull/201. These changes are more for developers than CASPER users who will use the tutorials_devel repo rather.