casper-astro/mlib_devel #202: update_casper_blocks doesn't seem to be permanent

kiranshila, 2023-07-20:
So, I have an m2019 model I'm trying to update to the m2021 toolflow.

 I update my mlib_devel submodule, run `./startsg`, open my model, and run `update_casper_blocks(bdroot)`, let it do its thing (updating 48 blocks), save the simulink model, and close matlab.

When I open things up again, and try to run the same `update_casper_blocks` it still seems to think there are 48 blocks to update. I'm not sure what to make of this.

jkocz, 2023-08-03:
It should always attempt to update all the blocks in the design that can be updated. Could it be that it's not just trying to re-update blocks that don't need it (ie the first update_casper_blocks was successful, and it was saved, it's just re-updating because it doesn't know better)? Or it is the case that the design actively doesn't work unless you do an update blocks each time?

jack-h, 2023-08-04:
Yeah, this is correct -- update_casper_blocks knows nothing about whether
blocks need to be updated, it just nukes and replaces them all.

kiranshila, 2023-08-04:
How would I know if blocks needed to be updated?

jack-h, 2023-08-04:
Not really a satisfying answer, but you'll know they _don't_ need to be
updated if you haven't updated your mlib_devel.

closed 2024-05-11 by jkocz