casper-astro/mlib_devel #208: ValueError: Verification of write to '' at offset 0 failed.

ywj0225, 2024-05-09:
I followed the red_pitaya tutorial one until executing the command "fpga.write_int('a',10)" and the problem started to appear.

In [5]: fpga.write_int('a',10)
2024-05-09 15:07:38.21 ERROR 192.168.31.206 casperfpga.py:547 - Verification of write to a at offset 0 failed. Wrote 0x0000000a... but got back 0x00000001.
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[5], line 1
----> 1 fpga.write_int('a',10)

File ~/work/cfpga_venv/lib/python3.8/site-packages/casperfpga/casperfpga.py:597, in CasperFpga.write_int(self, device_name, integer, blindwrite, word_offset)
    595     self.blindwrite(device_name, data, word_offset * 4)
    596 else:
--> 597     self.write(device_name, data, word_offset * 4)
    598 self.logger.debug('Write_int %8x to register %s at word offset %d '
    599                   'okay%s.' % (integer, device_name, word_offset,
    600                   ' (blind)' if blindwrite else ''))

File ~/work/cfpga_venv/lib/python3.8/site-packages/casperfpga/casperfpga.py:548, in CasperFpga.write(self, device_name, data, offset)
    543 err_str = 'Verification of write to %s at offset %d failed. ' \
    544           'Wrote 0x%08x... but got back 0x%08x.' % (
    545               device_name, offset,
    546               unpacked_wrdata, unpacked_rddata)
    547 self.logger.error(err_str)
--> 548 raise ValueError(err_str)

ValueError: Verification of write to a at offset 0 failed. Wrote 0x0000000a... but got back 0x00000001.

I  ues Red Pitaya OS 2.00-23 / STEMlab 125-14 LN v1.1

jkocz, 2024-05-10:
Hi @ywj0225,

What OS and Version are you using, and which casperfpga branch? 

Thanks.

ywj0225, 2024-05-11:
@jkocz Hi，I have vivado2021.1, matlabR2021a, python 3.8 on ubuntu20.04. And the version of mlib_devel is m2021a, the version of casperfpga is '0.4.4.dev1336+py38.276ee44'.

jkocz, 2024-05-13:
Thanks. I will try to replicate the issue, but it may take me some time to set things up.

Are you on the CASPER Slack/mailing list? I posted your question to the Red Pitaya slack channel. One suggestion was to "use the 1.04x versions of the Red Pitaya OS" which may be worth a try in the short term. 

Tagging a few of the people who did the RP dev in case they can comment (@bjbford @moragb96 @talonmyburgh) - It may be we need to recompile the RP image for the newer OS (@mitchburnett?)

wnew, 2024-05-13:
Hi @ywj0225

Firstly did the previous command in the tutorial return true or false? 
fpga.upload_to_ram_and_program('<your_fpgfile.fpg>')

If it returned true, did you definitely have software registers in your design?

Regards

Wes

ywj0225, 2024-05-14:
@jkocz After I reduced the the version of red_pitaya to 1.04x, everything was ok. Many thanks! 
 Today I tried tutorial 3Wide(-ish)band Spectrometer of red_pitaya, I changed the platform in tut_spec.slx to RED_PITAYA_14:xc7z010 and changed the yellow block red_pitaya_adc to 14 bits, resulting in an fpg file. But when I ran the command" python tut_spec.py -f <fpgfile name> -r <Red-Pitaya IP or hostname> -a <accumulation length>", the process was stuck and the result diagram cannot be obtained.

ywj0225, 2024-05-14:
@wnew Hi， they are both right. Thanks.

jkocz, 2024-05-14:
@ywj0225 Great! I'm glad that fixed the problem.

It sounds like your new issue is different, and probably a question better answered on the mailing list/slack.

Did you enter the command exactly as you have here, or did you include the fpg filename, IP address and accumulation length?

Did it say it was connection/programming/print the registers?

mitchburnett, 2024-05-15:
@ywj0225 please check and see if casper-astro/tutorials_devel#84 resolves your new issue with running the Red Pitaya tutorial spectrometer. I was able to reproduce the issue and was due to framing issues with the FFT block.

@jkocz we could recompile and be compatible with current RP base OS. Although, at the very least it looks like we need to update instructions with the compatible version.

closed 2024-06-01 by jkocz