casper-astro/mlib_devel #210: Guide to add new supported platform

TuringTW, 2024-05-26:
Hi,

We are developing an RF receiver using a board based on XCZU28DR-2FFVG1517. I am wondering if there's a guide on how to add a platform in Casper and how to create a Casperized image file for a specific board.

Thank you!

jack-h, 2024-05-27:
Have you seen the documentation at https://casper-toolflow.readthedocs.io/en/latest/jasper_documentation.html# , and in particular the [Supporting New Hardware](https://casper-toolflow.readthedocs.io/en/latest/jasper_documentation.html#supporting-new-hardware) section?

TuringTW, 2024-05-28:
Hi @jack-h 
Thanks for the quick response! Yes, I have read the document. But it doesn't cover preparing the image file for a new platform. Is there any suggestion on how to do it?